cmseaton42/node-ethernet-ip #4: Micro800 Support

patrickjmcd, 2018-03-14:
Provide support for accessing Micro800 PLCs

## Current Behavior
The system is currently unable to connect to a Micro800 PLC. The connection times out when attempting to access the PLC.

## Expected Behavior
Reading/writing data for Micro800 PLC's should work exactly the same as CLX and CPLX

## Possible Solution 
Determine what connection parameter changes need to occur and modify or extend the PLC class to accommodate changes.

## Context
This will broaden support for AB PLC's to include the Micro800 line of PLCs.

antoniocarranza, 2018-11-28:
Hello, I need to comunicate with Micro850 PLC, can I help in this issue?

cmseaton42, 2018-11-28:
Absolutely @antoniocarranza, we welcome all support. I think that there may be someone already working through getting this figured out as well.  I encourage you to join our gitter im group as well.

patrickjmcd, 2018-11-28:
I had been working on it, but I made almost 0 progress before my work schedule pulled me away from it entirely. I was working on comparing the packets being sent via the python package pycomm (works with micro800) to the packets being sent with this package. I'd encourage you to take over!

antoniocarranza, 2018-12-03:
Well, I'm willing but I do not know if I can help because I really do not know where to start. I installed pycomm I tried to communicate with the 850 but due to the poor documentation without success. Suppose I get it, should I use ethereal or similar to compare packages? Could you tell me what is the way of working?

antoniocarranza, 2018-12-03:
I have already communicated, in the source code was the solution. I continue with my question. What am I supposed to do now? Compare packages with ethereal?

manueldiasmanuel, 2019-03-03:
Any news on this topic ?   

Thanks,

MD

kyle-github, 2019-03-04:
When I implemented Micro800 support in my library, all I had to do was to leave out the path part.   It does not take an IOI path that includes the hops between the network module, the back plane and the CPU as in a  ControlLogix.  I ended up with code that was identical other than that.  Overall the Micro800 has a subset of the functionality of a ControlLogix.

There are a few other differences (from memory, it has been a few years):

- Looks like strings are not done the same way (one byte count plus data bytes rather than 4-byte count IIRC).
- it **requires** connected messages.   It does not support unconnected messages.
- I have not tried with larger packet sizes.  ControlLogix can do ~500 or ~4000 depending on the firmware and hardware.   I think Micro800 might be limited to ~500 bytes.
- I remember reading somewhere that Micro800 cannot bundle multiple CIP requests into one packet.   This is CIP function 0x0A.

punisher97, 2019-10-02:
Hi, i write my own package with support Micro800, I inpired in pylogix, written in python,  https://www.npmjs.com/package/node-logix 
I hope help to any :)

CryoBuilt, 2021-02-18:
Just curious if this has been worked on at all. We would love to have support for the Micro800 family. What was preventing this connection from working?