cmseaton42/node-ethernet-ip #5: Unable to read from Control Logix PLC

MattCost, 2018-03-27:
Unable to read anything from a control logic PLC.
Unable to display controller properties, unable to read a tag value.

## Current Behavior

When connecting to PLC, an UnhandledPromiseRejectionWarning is given.
I am able to ping the PLCs IP. (EN2TR card) I am able to go online with Studio 5000 programming software. 
PLC is in Run mode.

## Expected Behavior

Connection to PLC should complete, and plc properties would be written to console

## Possible Solution (Optional)

## Context

## Steps to Reproduce (for bugs only)

1. Download current version of Node and install. node-v9.9.0-x64 
2. Create a project directory and install node-ethernet-ip ("npm install ethernet-ip --save")
3. Create plc01.js file with sample connection code
4. Run node plc01.js
5. Receive error on console

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
ethernet-ip@1.1.0
dateformat@3.0.3
* Node Version (Use `node --version` - e.g. *9.8.0*):
v9.9.0
* Operating System and version:
Windows 7 - SP1 - x64
* Controller Type (eg 1756-L83E/B):
1756-L75
* Controller Firmware (eg 30.11):
28.012

4 slot rack, plc is in slot 0. EN2TR cards in slot 1 and slot 2. Slot 3 is open.

##Error output
(node:14792) UnhandledPromiseRejectionWarning: #<Object>
(node:14792) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:14792) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

##Test Code
const { Controller } = require("ethernet-ip");
const PLC = new Controller();
// Controller.connect(IP_ADDR[, SLOT])
// NOTE: SLOT = 0 (default) - 0 if CompactLogix
PLC.connect("10.115.45.55", 0).then(() => {
 console.log("Connected to PLC!");
 console.log(PLC.properties);
});

cmseaton42, 2018-03-27:
Hey man, sorry to hear about the issue.  If you have a few minutes, can you run the following code and let me know what it prints?

```js
const { Controller } = require("ethernet-ip");
const PLC = new Controller();

// Controller.connect(IP_ADDR[, SLOT])
// NOTE: SLOT = 0 (default) - 0 if CompactLogix
PLC.connect("10.115.45.55", 0).then(() => {
    console.log("Connected to PLC!");
    console.log(PLC.properties);
}).catch(e => {
    console.log(PLC.properties);
    console.log("=========================");
    console.log(e);
});
```

This should give some more insight into what might be happening.

MattCost, 2018-03-27:
C:\NodeJS\PLC01>node plc01.js
{ name: '1756-L75/B LOGIX5575',
  serial_number: 14878836,
  slot: 0,
  time: null,
  path: <Buffer 01 00>,
  version: '28.12',
  status: 12384,
  faulted: false,
  minorRecoverableFault: false,
  minorUnrecoverableFault: false,
  majorRecoverableFault: false,
  majorUnrecoverableFault: false,
  io_faulted: false }
=========================
{ generalStatusCode: 8, extendedStatus: [] }

Time Settings. Only issue is DST adjustment is off.
[image]

cmseaton42, 2018-03-27:
ahhhh, yep this is definitely related to #1.  The `readWallClock` method in the connect method is causing the issue, apparently this feature only works with the latest rockwell releases (e.g. L8 series controllers).  We will update the code soon and release a patch today.  @patrickjmcd or myself will update the thread once the changes are live. Thanks for helping to track this down =].

patrickjmcd, 2018-03-27:
You should be able to upgrade the npm package `npm install --save --upgrade ethernet-ip` and your issue should be resolved. The call to `readWallClock` has been removed so you should no longer get that error.

cmseaton42, 2018-03-27:
@MattCost `v1.1.1` has been released (thanks to @patrickjmcd).  Run `npm update ethernet-ip` to get the latest version.  After that, try running your code again and let us know if this resolves your issue.

cmseaton42, 2018-03-27:
@MattCost Can you confirm that `v1.1.1` resolves your issue?

MattCost, 2018-03-27:
Yup! All working now

cmseaton42, 2018-03-27:
Awesome! 🎉 Closing this.

closed 2018-03-27 by cmseaton42

RuanV, 2020-02-07:
HI Can Some one please help me I Have a Similar problem. But I keep getting this error:
caught error!Error Code:undefined { generalStatusCode: 5, extendedStatus: [] }.

Does this mean information is missing in the readtag parameters?
This is my Code:
var test = new Tag("CCW_Roll_RRCtrl", null, DINT);
PLC.connect("192.168.1.200", 2).then(async () => {
    var { name } = PLC.properties;

    console.log(`\n\nConnected to PLC ${name}...\n`);
    await PLC.readTag(test);
    console.log(tes.value);
}).catch(error => {
    console.error('caught error!' + 'Error Code:' + error.Code, error);
});

**Secondly:**
Does this module connect with the Allen bradley Rockwell 1756-EN2T/D\u0003\u0000\u0000\u0000 PLC

jhenson29, 2020-02-08:
1756-EN2T is an Ethernet card, not a PLC. You are connecting to slot 2. Is that where the EN2T is? Because that would be wrong. You need to enter the slot number for the processor, the one with a keyswitch and probably a 1756-Lxx part number. As a reminder, slot counting starts at 0.

RuanV, 2020-02-10:
Okay great, I did not take that in consideration. I will test today and see what is my results. Thanks for the assistance