cmseaton42/node-ethernet-ip #7: Changing PLC Scan Rate has no effect

MattCost, 2018-03-27:
Changing PLC Scan rate has no effect.
Tags are scanned as fast as possible

## Current Behavior

Tags are scanned as fast as possible

## Expected Behavior

Tags should be scanned every X ms

## Possible Solution (Optional)

ethernet-ip\src\controller\index.js
line 433 - should be "await delay(this.state.scan_rate);"

## Context

## Steps to Reproduce (for bugs only)

1.
2.
3.
4.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
* Node Version (Use `node --version` - e.g. *9.8.0*):
* Operating System and version:
* Controller Type (eg 1756-L83E/B):
* Controller Firmware (eg 30.11):

cmseaton42, 2018-03-27:
Hmmm, the master branch shows line 433 as what you are recommending already. Can you add your code so I can get a look at what you are doing?

MattCost, 2018-03-27:
https://github.com/cmseaton42/node-ethernet-ip/blob/master/src/controller/index.js
That version doesn't have the await

	const { Controller, Tag } = require("ethernet-ip");

	const PLC = new Controller();

	// Add some tags to group
	PLC.subscribe(new Tag("TestRead")); // Controller Scope Tag

	PLC.connect("10.115.45.52", 0).then(() => {
		console.log("Connected to PLC!");
		console.log(PLC.properties);
	 
		PLC.scan_rate = 1500;
		
		// Begin Scanning
		PLC.scan();
	});

	// Catch the Tag "Changed" and "Initialized" Events
	PLC.forEach(tag => {
		// Called on the First Successful Read from the Controller
		tag.on("Initialized", tag => {
			console.log("Initialized", tag.value);
		});

		// Called if Tag.controller_value changes
		tag.on("Changed", (tag, oldValue) => {
			console.log("Changed:", tag.value);
		});
	});

cmseaton42, 2018-03-27:
Ahhhh, good catch. Patching now.

cmseaton42, 2018-03-27:
@MattCost `v1.1.2` should resolve the issue. Let me know if everything seems to be working when you upgrade.

closed 2018-03-27 by cmseaton42

MattCost, 2018-03-27:
Yup! Working as expected now.

cmseaton42, 2018-03-27:
Sweet, Thanks for the heads up!

PR #8 (2018-03-27): Allows scan rate to work properly

Fixes bug discovered in #7 

## Types of changes

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to change)

## Checklist:

- [x] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have updated the documentation accordingly.
- [x] I have read the **CONTRIBUTING** document.
- [ ] I have added tests to cover my changes.
- [x] All new and existing test [... cut]
commit 615fc951 (2018-03-27): Fix #7 allows scan rate to work properly