cmseaton42/node-ethernet-ip #13: No support for tags or type STRING, ARRAY, or LINT.

cmseaton42, 2018-04-05:
## Current Behavior

Currently, there is no support for tags of type String, Array, or LINT. 

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*): latest
* Node Version (Use `node --version` - e.g. *9.8.0*): 9.10.0
* Operating System and version: win7x64
* Controller Type (eg 1756-L83E/B): n/a
* Controller Firmware (eg 30.11): n/a

jdingus, 2018-05-11:
I would be happy to do some shop floor testing on this once it's ready. - Thanks Josh

jhenson29, 2018-05-31:
I find that I can't read atomic array elements.  e.g. with Tag type DINT[2], I can't read "Tag[0]" or "Tag[1]".

Is that expected?

Is there an expected date for string/array/udt support?

cmseaton42, 2018-05-31:
@jhenson29 I have paused development for the moment as work is pretty insane for me. I hope to get back to full speed beginning early July which would make a potential release window of mid to late July. 

I apologize for the inconvenience. I will certainly update this thread when I begin adding the remaining *first class* functionality. If possible, I will work to get a release candidate out earlier than that. That said, pull requests are welcome and if you would like to take on the challenge to get something implemented sooner then I would appreciate the help 😄.

jhenson29, 2018-05-31:
No need to apologize. What you’ve done so far is awesome. I have some free time this weekend. I’ll look through it, but javascript is new to me (I’m coming primarily from C#) and I normally use a .net library for EIP. I have written my own code for plain tcp sockets on both ends though so I’ll have a go at it.

cmseaton42, 2018-05-31:
Thanks man!  It has been a fun project for me.  Feel free to join our [gitter](https://gitter.im/node-ethernet-ip/Lobby) chat if you have any questions about the code base.

cmseaton42, 2018-09-10:
I am picking up development of this functionality now.  Updates to come as I make progress!

seth350, 2018-11-28:
+5 for STRINGs! 😀

jjaramillom, 2019-11-11:
First of all, thanks for the awesome job @cmseaton42 . Just checking if the support of these types is still on the road map? Let me know if you could use some help :)

SerafinTech, 2019-11-16:
Just committed STRING and LINT support to my fork: https://github.com/SerafinTech/node-ethernet-ip (Tests still need to be written: USE WITH CAUTION!)

mataherry, 2020-09-25:
> Just committed STRING and LINT support to my fork: https://github.com/SerafinTech/node-ethernet-ip (Tests still need to be written: USE WITH CAUTION!)

Can we suggest adding the changes to main repo?