cmseaton42/node-ethernet-ip #17: while tag is "Ver.motoer.I" ,error Tag is not string

nuannuande, 2018-05-11:
src/tag.js
 const regex = /^[a-zA-Z_][a-zA-Z0-9_]*([a-zA-Z0-9_]|\[\d+\])$/i; 

## Current Behavior

## Expected Behavior

## Possible Solution (Optional)

## Context

## Steps to Reproduce (for bugs only)

1.
2.
3.
4.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
* Node Version (Use `node --version` - e.g. *9.8.0*):
* Operating System and version:
* Controller Type (eg 1756-L83E/B):
* Controller Firmware (eg 30.11):

cmseaton42, 2018-05-11:
Can you provide more details surrounding the issue you are having? Code snippet, error screenshot, etc. 

Thanks!

nuannuande, 2018-05-17:
F:\PLCgate>node -v
v8.9.3

F:\PLCgate>npm list
plcgate@1.0.0 F:\PLCgate
+-- ethernet-ip@1.1.6
| +-- dateformat@3.0.3
| `-- task-easy@0.2.0
+-- redis@2.8.0
| +-- double-ended-queue@2.1.0-0
| +-- redis-commands@1.3.5
| `-- redis-parser@2.6.0
`-- socket.io-client@2.1.0
  +-- backo2@1.0.2
  +-- base64-arraybuffer@0.1.5
  +-- component-bind@1.0.0
  +-- component-emitter@1.2.1
  +-- debug@3.1.0
  | `-- ms@2.0.0
  +-- engine.io-client@3.2.1
  | +-- component-emitter@1.2.1 deduped
  | +-- component-inherit@0.0.3
  | +-- debug@3.1.0 deduped
  | +-- engine.io-parser@2.1.2
  | | +-- after@0.8.2
  | | +-- arraybuffer.slice@0.0.7
  | | +-- base64-arraybuffer@0.1.5 deduped
  | | +-- blob@0.0.4
  | | `-- has-binary2@1.0.2 deduped
  | +-- has-cors@1.1.0 deduped
  | +-- indexof@0.0.1 deduped
  | +-- parseqs@0.0.5 deduped
  | +-- parseuri@0.0.5 deduped
  | +-- ws@3.3.3
  | | +-- async-limiter@1.0.0
  | | +-- safe-buffer@5.1.1
  | | `-- ultron@1.1.1
  | +-- xmlhttprequest-ssl@1.5.5
  | `-- yeast@0.1.2
  +-- has-binary2@1.0.2
  | `-- isarray@2.0.1
  +-- has-cors@1.1.0
  +-- indexof@0.0.1
  +-- object-component@0.0.3
  +-- parseqs@0.0.5
  | `-- better-assert@1.0.2
  |   `-- callsite@1.0.0
  +-- parseuri@0.0.5
  | `-- better-assert@1.0.2 deduped
  +-- socket.io-parser@3.2.0
  | +-- component-emitter@1.2.1 deduped
  | +-- debug@3.1.0 deduped
  | `-- isarray@2.0.1 deduped
  `-- to-array@0.1.4

code goes here

const { Controller, Tag ,TagGroup } = require("ethernet-ip");
const PLC = new Controller();
var postdata={data:{dName:"win7",dId:"win_7",iData:[]},gSN:"netgate001",accessToken:"fc1198178c3594bfdda3ca2996eb65cb"};
var postflag=0;
var idata=[];

PLC.subscribe(new Tag("VersaRoll.VRoll.I","S030")); // 
PLC.connect("178.18.10.21",0).then(() => {
    // Set Scan Rate of Subscription Group to 50 ms (defaults to 200 ms)
    PLC.scan_rate=50;//
    // Begin Scanning
    PLC.scan();
});
setInterval(()=>{
PLC.forEach(tag => {
 //if(!tag.value) tag.value=0;
 console.log(tag.value);
}); 
},100)

//setInterval(()=>{
// update_Data();
//},1000);
var http = require("http");
var querystring = require('querystring');

tag.js  /RegExp cannot match tha tag name while the last only is one letter

cmseaton42, 2018-05-24:
@nuannuande, @patrickjmcd released a patch to help you with your issue, have you been able to test it?

you should be able to update to the latest version with `npm update`

We still have an issue when referencing a boolean via bit level specification that will be addressed at a later time.

closed 2018-05-24 by cmseaton42