cmseaton42/node-ethernet-ip #22: is it possible to convert this code to C#?

wyddd, 2018-06-11:
## Current Behavior

## Expected Behavior

## Possible Solution (Optional)

## Context

## Steps to Reproduce (for bugs only)

1.
2.
3.
4.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
* Node Version (Use `node --version` - e.g. *9.8.0*):
* Operating System and version:
* Controller Type (eg 1756-L83E/B):
* Controller Firmware (eg 30.11):

cmseaton42, 2018-06-11:
Absolutely! 😄 

The implementation would be different though as javascript works in an asynchronous fashion by default thus the Socket IO was written in a way that takes advantage of what javascript does best. I imagine this would change in a C# implementation. That said, anything synchronous within the source (building message buffers, etc) could be easily adapted.  

Does this answer your question?

jhenson29, 2018-06-11:
C# has asynchronous programming. I'm not familiar enough with aync in C# (or javascript in general) to compare/contrast them, but C# does have `await` and `async` keywords.

cmseaton42, 2018-06-11:
@jhenson29 I did not know that (not a .net guy tbh). But if so then it may be easier to convert then I thought.  You would just need to find a good TCP/IP socket class to use as a base.

patrickjmcd, 2018-06-11:
Thats the exact issue I'm having with Swift right now... finding the right Socket class to extend

patrickjmcd, 2018-06-11:
But to help answer the question, yes, it should be possible, _but_ it's not in the scope of this project. If you'd like to start your own project and use this code as a reference, be sure to let us know and we'll follow along and help as we can.

cmseaton42, 2018-06-19:
closing this

closed 2018-06-19 by cmseaton42