cmseaton42/node-ethernet-ip #23: Can't find this module

Dhiaaeddinanabtawi, 2018-06-19:
I can't find it when I search for it in Manage palette. When downloading it using npm, it doesn't seem to be downloaded successfully?

cmseaton42, 2018-06-19:
How are you trying to install it? with `npm i ethernet-ip --save`?

Dhiaaeddinanabtawi, 2018-06-19:
Yeah, it seems to be downloaded successfully, but I can't see the module anywhere in Node-RED

cmseaton42, 2018-06-20:
This is a node-red module, it is a nodejs module.  I am largely unfamiliar with node-red, but I think one could certainly build a node-red module using `node-ethernet-ip`.

closed 2018-06-28 by cmseaton42

gfcittolin, 2018-06-28:
We're in progress of creating a Node-RED node for this, but it's in very early stage of development. We have already the [node-red-contrib-s7](https://github.com/netsmarttech/node-red-contrib-s7) and the [node-red-contrib-pccc](https://github.com/netsmarttech/node-red-contrib-pccc), and it'll work in a very similar way.
I'll post here when we have something public. Testing will be much appreciated.

cmseaton42, 2018-06-28:
Awesome! :grin:

gfcittolin, 2018-08-23:
Hi guys,
we've published the `node-red-contrib-cip-ethernet-ip` node, it's already available for installation through Node-RED's interface. Please let me know if you have any problems when using or if you have any suggestion, they'll be much appreciated :)

Dhiaaeddinanabtawi, 2018-12-03:
Is it working with Omron PLC's as well? It is written that it is mainly for
Allen Bradely PLC's?