cmseaton42/node-ethernet-ip #24: I added the ability to read strings by modifying 'parseReadMessageResponse(data)' method. I add this to the case/ switch:             case 672:                  this.controller_value = data.toString('ascii', 8);                 break;

tjusticeus, 2018-07-11:
## Current Behavior

## Expected Behavior

## Possible Solution (Optional)

## Context

## Steps to Reproduce (for bugs only)

1.
2.
3.
4.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
* Node Version (Use `node --version` - e.g. *9.8.0*):
* Operating System and version:
* Controller Type (eg 1756-L83E/B):
* Controller Firmware (eg 30.11):

jhenson29, 2018-07-13:
L83...High power!  We're actually actively working on integrating string read/write in a more general way along with UDTs right now. Come join us on gitter for more discussion!

austin557, 2019-07-18:
@tjusticeus  How exactly were you able to read strings? I really need the ability to just read strings but haven't been able to figure it out yet.

tjusticeus, 2019-07-18:
It has been awhile since i looked at this. I will have to refresh my memory. Get back to you soon.

tjusticeus, 2019-07-18:
go to ethernet-ip/src/tag/index.js

go to method 'parseReadMessageResponse(data)'    //Line 304

under the 'break' of case BOOL, I added another case

case 672:
this.controller_value = data.toString('ascii', 8);
break;

austin557, 2019-07-19:
@tjusticeus Thanks for the help!

RubenFarfanL, 2020-07-17:
any know how to write string?