cmseaton42/node-ethernet-ip #26: [FEATURE REQUEST] support tag enumeration

spike0xff, 2018-07-20:
I'm not very familiar with Ethernet/IP, CIP, etc. but I know that some Rockwell/AB PLCs can enumerate - send a list of - all the tags that are available, along with some information about each tag.  I know rslinx can do this. I'd like my stand-alone applications to be able to do it too.

## Current Behavior

## Expected Behavior
Seems like this would be a new Controller method that actively enumerates tags (maybe with some filtering options) and returns an array or object representing a tag collection.
In the simplest case, after connecting to a PLC, you'd call getTagList or getSymbolList(options) - it would return an object similar to a TagGroup but with more info per tag.

## Possible Solution (Optional)
Would be very happy to contribute code, if I can figure out the code and how to integrate it.

## Context
My company uses relatively lightweight edge devices running either Windows or Linux, and we distribute these to multiple customers.  We encounter a variety of PLCs running a wide variety of programs, some well documented, some not at all.  We prefer not to go on-site if we don't have to, and we don't want a complex or labor intensive setup process.  I'd prefer not to rely on (or install) bulky or platform-dependent tools (not naming names but rslinx and CCW).
I want the ability to 'browse' a PLC in the field with platform-independent node code, automatically, given only its IP address.

## Your Environment
* Operating System and version: Windows 7 and 10, Linux (stripped down Ubuntu).
* Controller Type (eg 1756-L83E/B): multiple, but often 1769's

jhenson29, 2018-07-20:
cmseaton42 was working on this as of around 3 weeks ago, but I don’t know what the status is. It’s also going to require finishing some additional items like UDT read/write (I’m currently working on, or was before I went on vacation this week) and fragmented read/writes. There’s a “manuals” folder in the project with info (specifically the two data access PDFs).

cmseaton42, 2018-09-10:
I am working on this now! Sorry for the delay :).  More to follow as progress is made!

spike0xff, 2018-09-28:
Hey that is awesome news, I hope to be in a position to work on ethernet-ip later in October!  I can help with testing at least ;-)

JollyWizard, 2019-07-04:
`libplctag` has an example in c that looks like it covers most of the details.

https://github.com/kyle-github/libplctag/blob/master/src/examples/list_tags.c.

SerafinTech, 2019-11-10:
Over a year since last update on this feature.  I added support for this in my fork.

https://github.com/SerafinTech/node-ethernet-ip/tree/v2.0.1

Enjoy!

kyle-github, 2019-11-10:
If you are using my code for examples, I recently added support for listing program tags too.   For other examples, please look at D.M Roeder's [pylogix](https://github.com/dmroeder/pylogix).   That's where I learned how to do it!

klmurphy72, 2020-06-01:
@SerafinTech ,

I am using your addition to the library successfully!  It is awesome.  I'm using it to pull all the tag data structure out and then search for instances of AOI's.  I then set internal tags of the AOI's.  This is a "name" tag that is the name of the enclosing tag.  This provides an Id that is provided externally instead of writing ladder logic to fill in the field.  A MAJOR time-saver!!

I see that you have added the ability to read / write string tags.  It does work, but not with user defined strings.  Is this to be expected?

Ken

SerafinTech, 2020-06-03:
@klmurphy72 ,

Here is a [link](https://youtu.be/-L3oSsjFYh0) to the first youtube video that I made to show how to use the library.  It shows how to read strings.  To write a string just set the tag.value = 'string' and then PLC.writeTag(tag).  I still need to update the README.

Jason

mataherry, 2020-09-28:
> Over a year since last update on this feature. I added support for this in my fork.
> 
> https://github.com/SerafinTech/node-ethernet-ip
> 
> Enjoy!

@SerafinTech is it possible to publish your palette to node-red repo :)?