cmseaton42/node-ethernet-ip #32: BOOL is always read as False

balex3, 2018-09-13:
## Current Behavior
`  PLC.connect("192.168.0.2", 0).then(async () => {
    const fooTag = new Tag("BoolTag");
    setInterval(async function(){
        await PLC.readTag(fooTag);
        console.log(fooTag);
    }, 2500);
});`

## Expected Behavior
Any BOOL tag I attempt to read is always false.

## Possible Solution (Optional)

## Context
Cannot read bool values, I have updated library.

## Steps to Reproduce (for bugs only)

1.
2.
3.
4.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*): 1.2.0
* Node Version (Use `node --version` - e.g. *9.8.0*): v10.0.0
* Operating System and version: Windows 7
* Controller Type (eg 1756-L83E/B): 1769-L33ERMS/A
* Controller Firmware (eg 30.11): 30.11

jhenson29, 2018-09-13:
Sorry about that. Earlier firmware versions return 0xff and later ones return 0x01. I'm not which version the change was made in. I know v20 is 0xff and v29 is 0x01.  I'm glad you were able to update and fix the issue.

cmseaton42, 2018-09-13:
Related to #29 

Closed with #30