cmseaton42/node-ethernet-ip #36: Tag.value is undefined after writing to PLC

jhenson29, 2018-09-26:
After writing a tag to the PLC, tag.value becomes undefined.

## Current Behavior

Tag value get set to undefined instead of the value it had been set to.

## Expected Behavior

Tag value should remain what it was set to.

## Possible Solution (Optional)

tag.unstageWriteRequest references this.state.controllerValue instead of this.state.tag.controllerValue
This is the source of the undefined value.

## Context

## Steps to Reproduce (for bugs only)

1. plc.writeTag(tag);
2. console.log(tag.value)

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
* Node Version (Use `node --version` - e.g. *9.8.0*):
* Operating System and version:
* Controller Type (eg 1756-L83E/B):
* Controller Firmware (eg 30.11):

jhenson29, 2018-10-03:
This still isn't right. I was doing more testing in response to the reports of tag writing issues. I fixed the `tag.controllerValue` reference in #37 but, I think there is another problem. `tag.controllerValue` isn't getting set so tag.value is just getting put set back to it's last read value. I think the assignment is backwards. I think maybe `tag.controllerValue` should be getting set to tag.value in the `unstageWriteRequest` method.

cmseaton42, 2018-10-03:
Ill wait for you to test the changes before closing this @jhenson29