cmseaton42/node-ethernet-ip #38: Add support for specifying non standard EtherNet/IP port number

jhenson29, 2018-10-02:
Add support for being able to add port number to IP address in Controller connect method.

## Current Behavior

Port number is set to 44818 and cannot be changed.

## Expected Behavior

Optionally specify port after IP address in connect method.
e.g.  `plc.connect('192.168.1.100:5000');`

## Possible Solution (Optional)

Parse connect IP address string and check for port number. Use port number passed for eip connect. Default to 44818 if no port provided (no change to current operation).

## Context

Mapping different ports on a WAN to different PLCs on a LAN.

## Steps to Reproduce (for bugs only)

1.
2.
3.
4.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
* Node Version (Use `node --version` - e.g. *9.8.0*):
* Operating System and version:
* Controller Type (eg 1756-L83E/B):
* Controller Firmware (eg 30.11):

patrickjmcd, 2018-10-02:
This would be really beneficial.

jhenson29, 2018-10-02:
It's just for the node end. I'm not aware of any way to change the port on the processor. So this is just for the specific use case of mapping ports on a router. But, yes, I've wanted to do this exact thing with other software numerous times and was not able to.

patrickjmcd, 2018-10-02:
I've had MANY conversations with rockwell support and firmware developers who all believe you _should_ be able to change the port from 44818, but none who are willing to push it forward.

jhenson29, 2018-10-02:
They believe correctly.

nikmartin, 2019-05-07:
just FYI there's many other PLCs with Ethernet/IP support that allow changing the port. Version 2.4 (April 
 2019) of Koyo Click PLCs now support Ethernet/IP and support changing the port.