cmseaton42/node-ethernet-ip #45: Question: is it possible to use without PLC

emielke12, 2018-12-12:
I have a Keyence Vision system that I am trying to grab data off of (just one result number for now). I am hoping to do this without needing to attach the keyence system to a PLC, and instead just grab the data directly from my Linux machine. Is this possible with this library? 

On the keyence side, looking at the EtherNet/IP configurations, it indicates that it is disconnected from the PLC, which I would expect, but I was hoping there was some way I could configure my Linux machine (running Ubuntu 16.04) to pull the data out without needing to also connect to the PLC (which we don't have, running this mostly for research purposes, not heavy-weight industrial use, yet)

Is this Possible?

cmseaton42, 2018-12-12:
The primary purpose of this API is to be used with a Rockwell PLC. That said, you may be able to use some of the lower level classes within the library (`CIP`, `ENIP`) as a base for getting something working.

bmhgit, 2018-12-18:
Not sure if I understand correctly, but I assume the vision system supports CIP explicit communication? If so, there must be some sort of datasheet, where the Object model is specified, hopefully including support for Get_Attribute_Single?
If the answer is still yes, I implemented a small patch that should be able to read every CIP Object (Class,Instance,Attr) via Get_Attribute_Single Service.
Can be found in the branch "generic-support".

emielke12, 2018-12-18:
@bmhgit I believe the vision system supports CIP communication, I'll poke around with your suggestion and see what I get. Did you possibly mean branch "port-support"? I only see 3 branches, master, port-support, and v2.0.0.

cmseaton42, 2019-01-11:
@emielke12, It will not work out of the box with a vision system.  The problem lies in multiple places. First the `Controller` class is written to accommodate specific objects within Rockwell Controllers thus it does not generalize to other platforms easily.  Second, the `Enip` class that the `Controller` class is built on utilizes explicit messaging by default.  If you can get access to the Vision System's CIP object references then you may be able to use the underlying generic [ethernet/ip](https://github.com/cmseaton42/node-ethernet-ip/tree/master/src/enip) API to read/write data to the system.