cmseaton42/node-ethernet-ip #46: Uncaught Index Out of Range Error

ajcanterbury, 2018-12-14:
## Current Behavior
RangeError: Index out of range (Most recent call first)
	at checkOffset (buffer.js line 977 col 11)
	at Uint8Array.Buffer.readUInt16LE (buffer.js line 1023 col 5)
	at Object.header.parse.buf [as parse] (/opt/losant/node_modules/ethernet-ip/src/enip/encapsulation/index.js line 227 col 26)
	commandCode: buf.readUInt16LE(0),
	at Controller._handleDataEvent (/opt/losant/node_modules/ethernet-ip/src/enip/index.js line 240 col 41)
	const encapsulatedData = header.parse(data);
	at emitOne (events.js line 116 col 13)
	at Controller.emit (events.js line 211 col 7)
	at addChunk (_stream_readable.js line 263 col 12)
	at readableAddChunk (_stream_readable.js line 250 col 11)
	at Controller.Readable.push (_stream_readable.js line 208 col 10)
	at TCP.onread (net.js line 601 col 20)

## Expected Behavior

## Possible Solution (Optional)
Add a check maybe.

## Context
Interrupts multiple reads. 

## Steps to Reproduce (for bugs only)
Random occurrence, not exactly sure how to reproduce (know that sucks). In a three second loop reading a Group of Tags.

## Your Environment
* Package version: yarn 1.12.3, npm 6.4.1
* Node Version: 10.14.2
* Operating System and version: Ubuntu 16.04 AMD64
* Controller Type (eg 1756-L83E/B): 1769-L16ER/B LOGIX5316ER
* Controller Firmware (eg 30.11): 31.11

ajcanterbury, 2018-12-14:
I don't know where this is coming from exactly. Most of the time everything works great. The error is random and not frequent.

I create a connection and then every three seconds I create a tag, read it, add that number of tags to a tag group, read it, then write to the tag group. Sort of like so:

```
// client.PLC Controller() and client.group TagGroup() is created the first time before the 3 sec loop
const strLen = new Tag('myTag.LEN', 'myProgram', DINT);
try {
  await client.PLC.readTag(strLen);
} catch (err) {
   throw err;
}

// first I clean out the old .DATA before adding the new to the group
for (let i=0; i<strLen[newTag].value; i++) {
  client.group.add(new Tag(`myTag.DATA[${i}]`, 'myProgram'));
}
// there are various other tags added to the group I'm leaving out
try {
  await client.PLC.readTagGroup(client.group);
} catch (err) {
  throw err;
}

// the writing is much more involved since data is coming from a separate source but for simplicity
const someText = 'Hello';
client.group.forEach((tag) => {
  charIndex = parseInt(tag.name.substring(tag.name.indexOf('.DATA[')+6, tag.name.length - 1), 10);
  tag.value = someText.charCodeAt(charIndex);
}

try {
  await client.PLC.writeTagGroup(client.group);
} catch (err) {
  throw err;
}
```

jhenson29, 2018-12-15:
This looks familiar. I was getting some weird error (I think this one) and I think it came down to group reads with a bad tag, but it was a while ago so I can’t be sure. 

I’m in the habit now of always reading tags individually first to check that they work, then only adding them to a group if they were successful on the individual read.

cmseaton42, 2019-01-11:
@ajcanterbury Did you double to check to make sure every tag in the group read existed?  If so, did this help resolve the issue?

cmseaton42, 2019-01-11:
hmmmm, so you would prefer an option to let the group fail silently in the case that a bad tag is detected.  Possibly auto-remove the bad tag from the group?  just thinking out loud

ajcanterbury, 2019-01-11:
Auto-remove could work if that's possible. Easy fix might be to throw an error for a bad buffer message and allowing the read to fail.

gfcittolin, 2019-01-13:
> hmmmm, so you would prefer an option to let the group fail silently in the case that a bad tag is detected. Possibly auto-remove the bad tag from the group? just thinking out loud

Something like this might be a good idea. The Node-RED node for this library currently have issues in this case, as if there's a single tag that cannot be read, the whole reading fail. I even have a script to read a list of them individually, so I can check before reading them in a group. 
Dropping them out of the group and emitting an event to signalize it would make it much easier :)