cmseaton42/node-ethernet-ip #47: plc5

gilesbradshaw, 2019-01-10:
would this work with plc5s?

jhenson29, 2019-01-11:
I don't think so. I think it only works with ControlLogix and CompactLogix right now. I didn't think PLC5 even had EIP, but I did a quick google search and saw that it does?

gilesbradshaw, 2019-01-11:
yes with a card you can bridge ethernet ip to dh+

cmseaton42, 2019-01-11:
That moment when you accidently hit the close issue button lol

cmseaton42, 2019-01-11:
Hey @gilesbradshaw, there currently isnt support for PLC5.  There is another library out in the node world that may help you that implements cpppo that may be of use to you but I am unable to locate it at the moment :(.  The library is written to reference the specific EIP Tag Instances in the more recent Rockwell controllers.

gilesbradshaw, 2019-01-11:
Ok thanks - maybe I have a play with it anyhow - I wonder how hard it could be to adapt

jhenson29, 2019-01-11:
This? https://www.npmjs.com/package/nodepccc