cmseaton42/node-ethernet-ip #54: Custom Queue length

Mudrekh, 2019-08-05:
## Current Behavior
The current limit for any given queue is the default defined in task-easy

## Expected Behavior
Pass any queue length so projects controlling a larger set of data can work. 

## Possible Solution (Optional)
Define opts in controller constructor and instantiate queues with the passed in parameter.

## Context
We are trying to poll larger subset of data individually, and would like to be able to add more than 100 points to the task queue.

cmseaton42, 2019-08-07:
Hmmm, this seems pretty reasonable. Feel free to submit a PR to add a custom queue size.

Mudrekh, 2019-09-04:
@cmseaton42 Im going to close this issue, but could you release an npm version? Would love to start using the npm verison of the package.

closed 2019-09-04 by Mudrekh

cmseaton42, 2019-09-04:
`v1.2.5` published