cmseaton42/node-ethernet-ip #56: READING ERRORS

guillermoLeon26, 2019-09-10:
I am trying to read DINT values, but sometimes these values ​​are incorrect, how can I know if a value I am reading is incorrect?

I want to read values ​​that are stored in the plc. These values ​​are voltage and energy. The voltage values ​​range from 427 to 429, but suddenly they go to 16187392 and again go back to normal. This seems to me a reading error

* Package version (Use `npm list` - e.g. *1.0.6*): 6.9.0
* Node Version (Use `node --version` - e.g. *9.8.0*):10.16.3
* Operating System and version: Linux 4.15.0-60-generic #67-Ubuntu  x86_64
* Controller Type:1768-L43
* Controller Firmware:20.14

cmseaton42, 2019-09-10:
Can you post a snippet of your code?

guillermoLeon26, 2019-09-10:
const { Controller, Tag } = require('ethernet-ip')
const fs = require('graceful-fs')

const delay = require('delay')

const PLC = new Controller()

const voltajeLineaAN = new Tag('TDLOW_PQMII.VOL_VAN')
const voltajeLineaBN = new Tag('TDLOW_PQMII.VOL_VBN')
const voltajeLineaCN = new Tag('TDLOW_PQMII.VOL_VCN')

const voltajeFaseAB = new Tag('TDLOW_PQMII.VOL_VAB')
const voltajeFaseBC = new Tag('TDLOW_PQMII.VOL_VBC')
const voltajeFaseCA = new Tag('TDLOW_PQMII.VOL_VCA')

const energiaReal = new Tag('TDLOW_PQMII.ENY_POS_REAL')

console.log('Tablero - TD_LOW_1')

const fecha = new Date().toISOString()
const fila = `VOLTEJE-AB;VOTEJE-BC;VOLTEJE-CA;VOLTAJE-AN;VOLTEJE-BN;VOLTAJE-CN;ENERGIA\n`

fs.appendFileSync(`${__dirname}/archivos/${fecha}.txt`,fila)

PLC.connect('172.30.0.85', 0).then(async () => {
  while (true) {
    try {
      await PLC.readTag(voltajeLineaAN)
      await PLC.readTag(voltajeLineaBN)
      await PLC.readTag(voltajeLineaCN)
      await PLC.readTag(voltajeFaseAB)
      await PLC.readTag(voltajeFaseBC)
      await PLC.readTag(voltajeFaseCA)
  
      await PLC.readTag(energiaReal)

      let fila = `${voltajeFaseAB.value};${voltajeFaseBC.value};${voltajeFaseCA.value};${voltajeLineaAN.value};${voltajeLineaBN.value};${voltajeLineaCN.value};${energiaReal.value}\n`

      await fs.appendFile(`${__dirname}/archivos/${fecha}.txt`, fila)

      // console.log('Voltaje AN', voltajeLineaAN.value)
      // console.log('Voltaje BN', voltajeLineaBN.value)
      // console.log('Voltaje CN', voltajeLineaCN.value)
      // console.log('Voltaje AB', voltajeFaseAB.value)
      // console.log('Voltaje BC', voltajeFaseBC.value)
      // console.log('Voltaje CA', voltajeFaseCA.value)
  
      // console.log('Energia Real', energiaReal.value)
    } catch (error) {
      console.log('error', error) 
    }

    await delay(100)
  }
})

jhenson29, 2019-09-11:
Have you verified that the values in the PLC aren’t actually doing this?

guillermoLeon26, 2019-09-11:
I am comparing the values with those that scada cimplicity throws at me, and that does not happen there.

Maybe it's because the scada cimplicity is already getting information

cmseaton42, 2019-09-11:
does the spike happen consistently, say after every 8th scan? or is it random?

cmseaton42, 2019-09-11:
Also, I recommend doing the following as it will limit the number of payloads to the PLC though I don't think it will fix your issue.

```javascript
const { Controller, Tag, TagGroup } = require("ethernet-ip");
const fs = require("graceful-fs");

const delay = require("delay");

const PLC = new Controller();

const voltajeLineaAN = new Tag("TDLOW_PQMII.VOL_VAN");
const voltajeLineaBN = new Tag("TDLOW_PQMII.VOL_VBN");
const voltajeLineaCN = new Tag("TDLOW_PQMII.VOL_VCN");

const voltajeFaseAB = new Tag("TDLOW_PQMII.VOL_VAB");
const voltajeFaseBC = new Tag("TDLOW_PQMII.VOL_VBC");
const voltajeFaseCA = new Tag("TDLOW_PQMII.VOL_VCA");

const energiaReal = new Tag("TDLOW_PQMII.ENY_POS_REAL");

// Use Tag Group
const group = new TagGroup();
group.add(voltajeLineaAN);
group.add(voltajeLineaBN);
group.add(voltajeLineaCN);
group.add(voltajeFaseAB);
group.add(voltajeFaseBC);
group.add(voltajeFaseCA);
group.add(energiaReal);

console.log("Tablero - TD_LOW_1");

const fecha = new Date().toISOString();
const fila = `VOLTEJE-AB;VOTEJE-BC;VOLTEJE-CA;VOLTAJE-AN;VOLTEJE-BN;VOLTAJE-CN;ENERGIA\n`;

fs.appendFileSync(`${__dirname}/archivos/${fecha}.txt`, fila);

PLC.connect("172.30.0.85", 0).then(async () => {
    while (true) {
        try {
            await PLC.readTagGroup(group);

            let fila = `${voltajeFaseAB.value};${voltajeFaseBC.value};${voltajeFaseCA.value};${
                voltajeLineaAN.value
            };${voltajeLineaBN.value};${voltajeLineaCN.value};${energiaReal.value}\n`;

            await fs.appendFile(`${__dirname}/archivos/${fecha}.txt`, fila);
        } catch (error) {
            console.log("error", error);
        }

        await delay(100);
    }
});
```

kyle-github, 2019-09-12:
The value you get is very suspicious: 16187392r10 = F70000r16.  The underlying protocol just reads the bytes in the tag.   I am not familiar with how node-ethernet-ip is handling those.   JavaScript is not the most fun language in which to handle binary data.

Based on the value in hex, I would be looking at the PLC program or the sensors first.

guillermoLeon26, 2019-09-12:
Thanks for answering

The peak occurs randomly and of varying magnitude. And yes I scan at the same speed as simplicity, but when I slow down the scan the same thing happens

I'm going to try another plc with a slower scan speed, I think that plc has problems because by checking I found that there are some memory overflows but I don't know exactly where they are, I'm going to keep investigating

guillermoLeon26, 2019-09-19:
The incorrect values were produced by a fault in the sensor that is connected to the PLC, now it works perfectly, I am also testing it in another PLC and so far everything works fine

cmseaton42, 2019-09-19:
Awesome, glad you got it figured out.  I'm gonna go ahead and close this issue.  :)

closed 2019-09-19 by cmseaton42