cmseaton42/node-ethernet-ip #58: TimeOut Error

Ramo38, 2019-10-21:
Hi Everbody;
I am trying to read tags from Allen Bradley Micro 850 PLC with node-red-contrib-cip-ethernet-ip node
but I am taking this error : Error connecting to PLC: Error: TIMEOUT occurred while reading Controller Props. How Can I solve this problem.

cmseaton42, 2019-10-23:
Thats interesting.  Can you post a snippet of your code?

jhenson29, 2019-10-23:
This package doesn’t support Micro800 processors yet, does it?

cmseaton42, 2019-10-23:
see issue #4, you are correct. It seems like once connected messaging is released as a first class feature in the ts implementation, we will be capable of supporting it but we cant as of this moment.  It is definitely something to keep on our radar though. Sorry @Ramo38