cmseaton42/node-ethernet-ip #62: Reading tags-UDTs

carloscheli, 2020-01-10:
Good morning
I have read on the npm website that you will soon be able to read tags for UDTs.
My question is when this functionality will be active or if there is at this time any beta version that includes this.
My question is when this functionality will be active or if there is at this time any beta version that includes this.
Thanks for your help

jhenson29, 2020-02-08:
PR https://github.com/cmseaton42/node-ethernet-ip/pull/25 has the code for reading UDTs.

jschenkDD, 2020-06-16:
Hi @jhenson29 i tried your branch with udt-templates. Reading tags work's fine, but writing udt's causes error:
Error reading tags { generalStatusCode: 31, extendedStatus: [ 258, 4099 ] }
=========================
{ generalStatusCode: 31, extendedStatus: [ 258, 4099 ] }

Do you have any idea?
Thanks for your help.

jhenson29, 2020-06-16:
I don't, off the top of my head. I have a test suite on my laptop that runs through reading/writing different types of tags, including UDTs (including deeply nested UDTs) against a real PLC. So, I think it should work in general. But I'm only _reading_ UDTs in my production applications. I'm not writing to any, So, I don't have much testing outside of my test suite. A few things.
1. I'll see if I can look into those extended status codes.
2. Does the tag your are trying to write to allow writing? (later versions of these processors allowed setting tags to be read only).
3. Can you share a copy of your UDT for testing?

jschenkDD, 2020-06-16:
2. Yes with other tools (TagMonitor..) tag could be written
Also if i define a tag like this:
```
const S_P1SW1_Flow = new Tag('S_P1SW1.Flow');
S_P1SW1_Flow.value = 20;
await PLC.writeTag(S_P1SW1_Flow);
```
But when i will write tag like this it doesn't work:
```
const S_P1SW1 = new Tag('S_P1SW1', null, 'S_P1SW1');
S_P1SW1.value.flow = 20;
await PLC.writeTag(S_P1SW1_Flow);
```

Template is:
```
plc.addTemplate({
            name: "S_P1SW1",
            definition: {
                flow: Types.Types.REAL,
                temperature: Types.Types.REAL,
                timeout: Types.Types.TIME,
            }
        });
```

jhenson29, 2020-06-16:
Did you try reading the tag first, and then writing? I'm not sure without going back through the code again, but a UDT tag's value structure may not be built correctly until after it is read once. 

Alternatively, you may try setting all of the values in the UDT manually before writing. 

Know that if you are writing a UDT tag like this, it will write ALL value in the UDT, not just the ones you set. It is read and written as a structure.

jschenkDD, 2020-06-16:
Hey,

yes i read first. Values read all successful, also setting all members causes this error

jschenkDD, 2020-06-18:
Do you know what error codes mean?