cmseaton42/node-ethernet-ip #76: Question: Can I get tag descriptions?

pilotkid, 2020-09-06:
In newer PLCs (version >20) there is a description field in RXLogics, that saves to the PLC. Is there a way to retrieve the tag comments/description with this library?

***Note:*** I may interchange in this issue the words description and comment, as we refer to them as comments but in RSLogix it is called description
 
## Current Behavior

On the tag properties, currently, there is not a description field. That would contain a comment/name for the tag. 

## Expected Behavior
When querying a tag, it would have a field property for the tag description. 

## Context
Most of the PLCs I work with nowadays have tag comments on almost all of the tags. When using this library sometimes I would need to see what the comment is for that tag. Currently, I have to find the tag and then hunt it down in RSLogix to find the description. However, when I am in the field this can be quite cumbersome.

jhenson29, 2020-09-06:
I know the current version doesn’t support it. It only returns values. 

There had been some work to read tags out of the processor, but I’m not sure if the descriptions are returned with that data or not. It may just be tag names and data structures. In any case, that hasn’t been merged into this version.

I’m not sure if any of the PRs or forks have a working version or not of reading other data out of the processor or not.

jhenson29, 2020-09-06:
I went back and looked at screen shots of the returned data for the tag and udt reads from another person on the Gitter feed and I didn’t see any meta data like comments returned.

pilotkid, 2020-09-07:
Ah dang, would it be possible to sponsor a feature like that? I am not at all familiar with the ethernet IP protocol and am not sure where to get started in order to make that a reality.