cmseaton42/node-ethernet-ip #79: Error: TIMEOUT occurred while reading Controller Props. - Micrologix 1100

paolodenti, 2020-10-27:
I cannot connect to a Micrologix 1100

## Current Behavior
Any operation after connect fails. Failure happens in `connect.then()` (therefore apparently upon successful connection).

Message is 
`Error: TIMEOUT occurred while reading Controller Props.`
happening in
`await this.readControllerProps()`
on controller/index.js, line 125

## Steps to Reproduce
```
const { Controller } = require("ethernet-ip");

const PLC = new Controller();

PLC.connect("192.168.150.44", 0)
    .then(() => {
        console.log(PLC.properties);
    })
    .catch((error) => {
        console.log(error);
    });

```

## Environment
ethernet-ip@1.2.5
Node Version v14.13.1
MacOS
Micrologix 1100 1763 L16AWA B/14.00

jhenson29, 2020-10-28:
Micrologix isn’t supported with this library at this time. Only Compact and ControlLogix processors.

paolodenti, 2020-10-28:
@jhenson29 
Thank you. My bad of course but I would suggest to explicitly write somewhere "No Micrologix" in the README.

Closing the issue.

closed 2020-10-28 by paolodenti

jhenson29, 2020-10-28:
The top of the readme says:

> A Lightweight Ethernet/IP API written to interface with Rockwell ControlLogix/CompactLogix Controllers.

Also, this is an EtherNet/IP library. EtherNet/IP is a specific application layer protocol which I don’t think is used by MicroLogix. I believe MicroLogix uses a different protocol similar to the SLC 5/05.

jhenson29, 2020-10-28:
You may want to look at the below package. I haven’t used it, so I can’t comment on how well it works. 

https://www.npmjs.com/package/nodepccc

paolodenti, 2020-10-28:
Thanks, I tried it yesterday, just a connection to the micrologix 1100, and it works flawlessly. I will update here after I will really try it heavily in the next days