cmseaton42/node-ethernet-ip #81: Error connecting to PLC: Error: TIMEOUT occurred while reading Controller Props.

ksar-ksar, 2021-02-23:
Hello,

I try to connect to a PLC that has a Duagon CIP Stack embedded.

When I use Schneider PLC, everything works.
Then I tried with Node-Red ethernet-ip and get a : "Error connecting to PLC: Error: TIMEOUT occurred while reading Controller Props."

Doing a bit of Wireshark on it, and below the extract of a working Schneider Communication : 
[image]

Then the Node-Red one : 
[image]

So the main difference between the two is that node-red request for a service Code 0x52 instead of 0x54 for the Schneider one.
Wireshark identifies this two services as : 
 * Schneider : 0x54 : Service: Forward Open (Request)
 * Node-Red : 0x52 : Service: Unconnected Send (0x52)

Do you know why node-red is using this 0x52 instead of 0x54 ? 
And is it possible to change it ?

Thanks for your help !

SerafinTech, 2021-02-23:
Looks like Node-red is using a version of ethernet-ip that only has unconnected messaging.  Schneider PLC looks like it is using connected messaging.  You could try updating the ethernet-ip library to the latest commit on GitHub.  That should support connected messaging.

ksar-ksar, 2021-02-23:
Hi,

Thanks for your answer. 
`npm view ethernet-ip version ` gives 1.2.5, that seems to be the last one

Alex-OPTIM, 2021-03-25:
Look at line [148 at src/controller/index.js](https://github.com/cmseaton42/node-ethernet-ip/blob/a5788f3470e4d5c61b8e4b4645d8922e23ee6e6b/src/controller/index.js#L148). It's still in TODOs.

If you would like to contribute and develop a `connected-send` module similar to [unconnected-send](https://github.com/cmseaton42/node-ethernet-ip/blob/master/src/enip/cip/unconnected-send/index.js) that would be great.