cmseaton42/node-ethernet-ip #87: Is this Project Still Maintained?

xinthose, 2021-06-17:
Hello,

   I am thinking of using this project in an industrial automation environment, so it's important it works properly.  I see that the last commit was on Sept 3rd, 2019 with 15 open pull requests.  Is this project still maintained by anyone?

cmseaton42, 2021-06-17:
Hey,

The project works well and has been used in several places around the world with great success.  I began doing a typescript rewrite about a year ago but have not yet completed it.  I do not have the time available at this point to fully support the project unfortunately though I think you will be able to use it with success if you choose to.  If you have any questions, feel free reach out and Ill do my best to anwser them.

closed 2021-06-17 by cmseaton42

xinthose, 2021-06-17:
@cmseaton42 OK.  Great.  I plan on using this in an express TypeScript project.

jhenson29, 2021-06-18:
@xinthose I’m actively using a version from this pull request: 

https://github.com/cmseaton42/node-ethernet-ip/pull/25

The only difference is support for UDT, which I need. Otherwise, the standard library works fine. I think everyone just got busy and stuff hasn’t been merged in. 

For myself, once the library was working with enough features, I moved onto things that *use* the library. 🙂

I’ve had this running in lots of production environments for two years, with new ones added all of the time. Zero issues. 

I also use with TypeScript.

xinthose, 2021-06-18:
@jhenson29 excellent; thank you for the feedback gentlemen