cmseaton42/node-ethernet-ip #89: Possible for multiple scan rates without creating a "new" controller for each one?

skykep, 2021-09-09:
## Current Behavior

## Expected Behavior

## Possible Solution (Optional)

## Context

## Steps to Reproduce (for bugs only)

1.
2.
3.
4.

## Your Environment

* Package version (Use `npm list` - e.g. *1.0.6*):
* Node Version (Use `node --version` - e.g. *9.8.0*):
* Operating System and version:
* Controller Type (eg 1756-L83E/B):
* Controller Firmware (eg 30.11):

pcnate, 2021-09-09:
I would say yes. Write your own function that performs a read based upon the scan rate timeout you wish to have.

skykep, 2021-09-09:
I meant natively rather than using setInterval to perform individual reads, which is less efficient than subscriptions.

pcnate, 2021-09-10:
I wouldn't use a setInterval. track the last time each tag was read and then get the oldest one most needing to be read. Add several to a tag group and read the tag group is also an option. A feature request for this library may be to add a scan rate to tag groups.

```pseudo js code

while( true ) {
  await getNextTagThatIsAgedEnough()
  await readThatTag()
  await extraDelayWhenNoTags()
}
```

skykep, 2021-09-10:
> A feature request for this library may be to add a scan rate to tag groups.

Great idea.  Appreciate your insight.

jhenson29, 2021-09-11:
I always just read tag groups in a loop with a delay, similar to what Nate shows. But I do end up setting up different controllers for each tag group and even run each controller on its own server. Usually 4 total for my standard HMI framework. But some of that may be specific for my application.